Loic74650/PoolMaster #1: Ajout Wifi NodeMCU V3 Esp8266

Cheveux, 2020-01-17:
très bon projet que je souhaite mettre en place chez moi mais étant tout nouveau dans ce domaine, j'essaye depuis quelques jours de remplacer la Shield ethernet par une nodeMcu V3 a base d'ESP8266 (ESP-12E).
Si tu as une idée sur comment flasher l'ESP pour pouvoir l'utiliser comme un Shield.

Loic74650, 2020-01-17:
Hello Cheveux

 

Désolé je n’ai pas d’expérience avec ESP

 

Loic

Cheveux, 2020-01-18:
ok merci pour l'info penses tu que si je prend un Arduino wifi R3 atmega2560 (a base de esp8266) je pourrais facilement adapter le projet ? j'ai besoin de passer par wifi car je ne souhaite pas passer par CPL dans mon local

Loic74650, 2020-01-18:
A partir du moment où tu restes sur des micro compatibles avec l’IDE Arduino, ça devrait aller. Par contre commence par passer en revue les librairies du projet PoolMaster et vérifie qu’elles prennent en charge le micro vers lequel tu veux migrer

closed 2020-02-05 by Loic74650