Loic74650/PoolMaster #5: {"PSIHigh":1.0}  upper or lower limit?

bbwouters, 2020-05-25:
{"PSIHigh":1.0}                  -> set the water high-pressure threshold (1.0bar in this example). When water pressure is over that threshold, an error flag is set.

Could it be the PSIHigh sets the lower limit for the filtration pump to run in stead of the upper limit?

 //If filtration pump has been running for over 7secs but pressure is still low, stop the filtration pump, something is wrong, set error flag 
    if(FiltrationPump.IsRunning() && ((millis() - FiltrationPump.LastStartTime)>7000) && (storage.PSIValue < storage.PSI_MedThreshold))
    {
      FiltrationPump.Stop();

Setting it at 1 bar in my cofig happily runs the pump generating a constant  5 bar.

I think the check is correct "if you see no pressure buildup there is a problem and stop the pomp" but the description is incorrect.

Loic74650, 2020-05-25:
I think all is correct, BUT:

 

*	storage.PSI_MedThreshold Is the variable used to detect a low pressure (0.25b by default in EEPROM, and no API function to change it)
*	storage.PSI_HighThreshold Is the variable to detect a high pressure (sand filter needs cleaning) but the error detection / flag is not implemented

bbwouters, 2020-05-25:
OK that explains why I do not get an error at 5 bar with 1.0 bar threshold. 

Sorry for creating confusion.

closed 2020-05-27 by Loic74650