Loic74650/PoolMaster #19: {"DelayPID":60}  when re-start during filter pump window

bbwouters, 2020-05-31:
Are we sure if the system is re-booted during the Flirtation pump window/ hours the PID regulation is delayed and does not start immediately regulating based on water left in the tubes (and therefore not reflecting real water quality?  Same question is valid when pump is manually stopped and started  during filtration hours. (As a matter of safety).

So does PHRegulationCallback(Task* me) checks when Firltration pump was last started? I don't think so but your insight would be appreciated.

Loic74650, 2020-05-31:
Correct it does not. 

I'd like the PHRegulationCallback(Task* me) to remain as clean and simple as possible, so I suggest that any future checks should be either performed in the generic loop or in the lower level Pump class.

In this particular case the simplest way to do this I think, is to reset the FiltrationPump.UpTime property after a manual stop/restart of the filtration. That way it will wait for another DelayPID before it regulates again. The downside is that it will messup the filtration real duration statistic for that particular day (maybe not so critical). 

A slightly more correct/complex way of doing things would be to increase the value of storage.DelayPIDs variable by the FiltrationPump.UpTime value after each start/stop of the filtration:

storage.DelayPIDs += FiltrationPump.UpTime

That way when it restarts, it has to wait for another while before starting the regulation again. 
At midnight, storage.DelayPIDs should be reset to its default value.

We also have to discuss whether EmergencyStopFiltPump should be reset at midnight or not (ie. should the double-tap feature be used for long-term filtration management or just small daily tasks)

bbwouters, 2020-05-31:
In my opinion: EmergencyStopFiltPump should be reset at midnight , "emergency" should the double-tap feature be used for small daily tasks, but a warning in the documentation it will restart next day.

Loic74650, 2020-05-31:
`In my opinion: EmergencyStopFiltPump should be reset at midnight , "emergency" should the double-tap feature be used for small daily tasks, but a warning in the documentation it will restart next day.`

I agree and will make the change

bbwouters, 2020-05-31:
unsigned long LastStartTime; 

can also be used to make a check with to take into account PID delay 

I'll have a look

Loic74650, 2020-06-01:
`unsigned long LastStartTime; can also be used to make a check with to take into account PID delay`

Correct. LastStartTime could be used to check how long the filtration was stopped for. If longer than, say, 1 hour, a delay should be applied before restarting the PIDs

Loic74650, 2020-06-02:
I gave some thoughts to this issue and how to handle it. 
It becomes unecessarily complicated once you wish to handle several double-taps in a row. 

Therefore v5.0.0 is now using a simple check before it starts the PIDS, I have replaced the following condition:

(FiltrationPump.UpTime/1000/60 > storage.DelayPIDs)

with:

(FiltrationPump.LastStartTime / 1000 / 60 > storage.DelayPIDs)

It has the inconvenience to enforce a DelayPID every time you stop/restart the filtration but in practice that should not be abig issue and that keeps things simple

bbwouters, 2020-06-02:
Perfect solution!

closed 2020-06-02 by Loic74650

Loic74650, 2020-06-16:
Hi Boris

 

Found the bug with DelayPID behaviour and the use of LastStartTime

I fixed it

 

Cheers

Loic

bbwouters, 2020-06-16:
Perfect!