Loic74650/PoolMaster #27: Nextion variable update missing

Gixy31, 2020-12-30:
When updating Filtration Start/Stop hours, the page0.vaStaSto.txt should be set with the new string in order to force updating of the field on all pages.
add line:
myNex.writeStr(F("page0.vaStaSto.txt"), temp);

Loic74650, 2020-12-30:
Thanks Gixy31, I believe I fixed that issue in my December 1st update of the develop branch, kindly confirm

Gixy31, 2020-12-30:
You're right! I was using the master branch.
Another fix in your pump library: on many relay boards, the relays are activated when pulling the input LOW ; to cope with that, you must not only invert the PUMP_ON, PUMP_OFF, INTERLOCK_OK and INTERLOCK_NOK constants, but you must also change the _isrunning_  and _Interlock_ functions like this:
return (digitalRead(isrunningsensorpin) == PUMP_ON);
return (digitalRead(interlockpin) == PUMP_ON);
And for the addtionnal relays commands, the digitalWrite instructions should also be inverted in the ProcessCommand() function.
All the best,
Denis

Loic74650, 2021-01-01:
Pump library: yes, nice catch
I think however that:

return (digitalRead(interlockpin) == PUMP_ON);

should be:

return (digitalRead(interlockpin) == INTERLOCK_OK);

Gixy31, 2021-01-01:
Exact, copiė collé malheureux

closed 2021-01-01 by Loic74650