Loic74650/PoolMaster #28: Compiling error

jdienlin, 2021-03-18:
Hi!
Really fascinated by your project! I have installed all libraries, using a Controllino Maxi board, however I receive the following error while compiling:

"Archiving built core (caching) in: C:\Users\jdienlin\AppData\Local\Temp\arduino_cache_542260\core\core_CONTROLLINO_Boards_avr_controllino_maxi_0c812875ac70eb4a9b385d8fb077f54c.a
C:\Users\jdienlin\Documents\Arduino\PoolMaster-master\PoolMaster-master\source\PoolMaster\PoolMaster.ino: In function 'simpLinReg':

PoolMaster:1876: error: unable to find a register to spill in class 'POINTER_REGS'

 }

 ^

PoolMaster:1876: error: this is the insn:

(insn 52 51 54 3 (set (reg:SF 67 [ D.2628 ])

        (mem:SF (post_inc:HI (reg:HI 82 [ ivtmp.77 ])) [12 MEM[base: _52, offset: 0B]+0 S4 A8])) C:\Users\jdienlin\Documents\Arduino\PoolMaster-master\PoolMaster-master\source\PoolMaster\PoolMaster.ino:1863 100 {*movsf}

     (expr_list:REG_INC (reg:HI 82 [ ivtmp.77 ])

        (nil)))

"

Any ideas?

Thanks!
Jonathan

Loic74650, 2021-03-18:
Hi
It's a bug from the compiler, you have to update the IDE and the arduino.avr-gcc

jdienlin, 2021-03-18:
Hi, thanks for the quick reply. Updating the compiler fixed the issue - big THANK YOU!

closed 2021-03-18 by jdienlin