Loic74650/PoolMaster #29: information

maury77, 2021-06-21:
I would need a simplified version for water monitoring.

It is possible to use it on

Arduino UNO R3 ATmega328P + WiFi ESP8266

arnolde, 2021-06-21:
I would think even an ESP8266 alone should do the trick... I'd be interested too, in case anybody gets into that.
I recently bought a PH-803W pool controller for 80€ on Aliexpress, it containes an ESP8266 too, but also a square microcontroller (type filed off). It's not bad, the app sucks, but I wrote a perl script to catch the UDP broadcasts of the ph+orp values and post them to an influxdb+grafana instance on my webserver. Other stuff includes Maxim ICL 7612D (for pH) and 7612A (for ORP) opamps and 2xTEL 6N137 optocouplers.

Loic74650, 2021-06-21:
PoolMAster was ported to ESP32 by @Gixy31, see here: https://github.com/Gixy31/ESP32-PoolMaster

closed 2021-07-11 by Loic74650