Loic74650/PoolMaster #35: Pump action is inverted

henjo, 2023-07-02:
Hi,
Thanks for a great project. 
I am trying to build a chlorine controller with a Controllino Maxi with the Poolmaster SW. 
At start the ORP and chlorine pump relays are all turned on and I have tracked down the problem to the definition of PUMP_ON macro in Pump.h. It is defined as active low which seems to be opposite to the way my Controllino hardware works. If the relay action is hardware dependent maybe it is good to have it as a parameter to the Pump class.

Loic74650, 2023-07-03:
Hi
Good point, will add it to the todo list!

Loic74650, 2023-07-04:
On a second thought, this issue could also come up with the other defines such as TANK_FULL/TANK_EMPTY and INTERLOCK_OK/INTERLOCK_NOK, but it would not be super elegant to have them all as parameters in the class constructor, would it? Any suggestion for a better way to do this?

henjo, 2023-07-05:
How about adding only one of the constants (PUMP_ON, TANK_FULL,
INTERLOCK_OK) as parameters to the constructor since they are binary
constants. So then there are 3 additional parameters to the constructor.
Not super elegant either. 

/Henrik 

2023-07-04 22:40 skrev Loic74650:

> On a second thought, this issue could also come up with the other defines such as TANK_FULL/TANK_EMPTY and INTERLOCK_OK/INTERLOCK_NOK, but it would not be super elegant to have them all as parameters in the class constructor, would it? Any suggestion for a better way to do this? 

  

Links:
------
[1]
https://github.com/Loic74650/PoolMaster/issues/35#issuecomment-1620724351
[2]
https://github.com/