Loic74650/PoolMaster #44: PH Temperature compensation

masterpj555, 2025-05-16:
Hi there Loic! Sorry for messaging you here as my knowledge of github is low.
 
Amazing project! I started attemtping to build my own using poolmaster.

I found some really good industrial PH sensors far cheaper then atlas scientific with temperature compensation
(140 with a controller, 89 without a controller compared to 319 from atlas scientific).

Now my question: Is it possible to add PH temperature compensation to poolmaster? It would increase accuracy.
I was under the impression the PH/ORP sensor board only does amplification and DAC.

closed 2025-05-16 by Loic74650

Loic74650, 2025-05-16:
+/-0.1 is probably maintained within that range without temp compensation. Whats more, reference solutions used to calibrate pH indicate the pH value to be used over that temperature range, so in a way it is already compensated for without actively doing it in the code, assuming the temperature stays relatively constant between calibrations. In the last resort, the function GetMeas() in the code shows how to use temp. compensation if really desired (it may need to be updated to work with the pH_Orp board)

masterpj555, 2025-05-16:
0.1 is plenty sufficient! I'm mostly wondering because i've seen some really eratic charts of scientific publications.. i have to check how far the drift is.. but in my country it (and so in france i believe) it can get quite cold and hot sporadically in the summer.

For example 15 degrees to sudden 28 degrees another day . I want to add wood fired and solar heated pool heating to my pool which means temps do tend to spike every now and then and not be consistent. if it despite that stays within 0.1 degrees then no problem but somehow I doubt it will.

masterpj555, 2025-05-16:
> pool

it may be wise to have redundancy anyway! in case of the sending a low cost temperature sensor can be added as a sort of low effort comparitor just for the pure sake of failure of a temperature sensor.. it wont be perfect but that redundancy can be used for error reporting. I do understand 2 probes is costly since it's a consumable.

arnolde, 2025-05-16:
> For example 15 degrees to sudden 28 degrees another day

Your pool water won't change temerature that fast. I have 20m³ and the temperature only sinks 1° or max 2° overnight.
During Covid I heated it to 32° in winter and it went down 4° overnight (freezing temps outside, covered the surface in 4cm styrofoam when not using) (not recommended btw, they soaked up the water)

masterpj555, 2025-05-16:
> > For example 15 degrees to sudden 28 degrees another day
> 
> Your pool water won't change temerature that fast. I have 20m³ (completely underground) and the temperature only sinks 1° or max 2° overnight. During Covid I heated it to 32° in winter and it went down 4° overnight (freezing temps outside, covered the surface in 4cm styrofoam when not using) (not recommended btw, they soaked up the water)

fair but since you want this system to sort of reduce your work on the pool the more hands off it is the better... say you could leave it more unattended over a period of 2~3 weeks youll have a fair bit of drift which will affect dosing.. possibly more chemicals are wasted in the process of adjusting back and forth.